cesarvandevelde/OctoPrint-M73Progress #2: Didn't works on PRUSA i3 MK2

bionno, 2018-01-18:


bionno, 2018-01-18:
HEllo, i've installed on Octoprint and connected to my printer PRUSA i3 MK2 but the progress did not shown .

Any clue to debug ?

cesarvandevelde, 2018-01-19:
Try manually entering `M73 P50` into the terminal and see if that updates the progress bar. If not, the issue is with your firmware. I took a quick glance at the Prusa Firmware repository, and I suspect it does not support the M73 command.

bionno, 2018-01-19:
The problem i think is inside the plugin because  tried another plugin and
that works fine.

cesarvandevelde, 2018-01-19:
As far as I'm aware, this is the only plugin that outputs `M73` commands to update the display's progress bar. Other plugins use `M117` (set LCD message) to output a progress message to the display. As noted in the plugin's readme, not all firmwares support the M73 command. If nothing happens when you enter `M73 P50` into the OctoPrint terminal, then that means your firmware does not support the command.

bionno, 2018-01-19:
Yes , sorry i read this too late, i'm using a plugin that push the M117
command .

Thanks again.
A.

closed 2018-01-19 by cesarvandevelde