cesarvandevelde/OctoPrint-M73Progress #3: Manual M73 command works but no progress during print

bobbanks, 2018-01-30:
Is any GCODE needed at the start of print?  I've confirmed LCD_SET_PROGRESS_MANUALLY is defined by manually setting the progress bar with an M73 in Terminal.  When printing from .gcode uploaded from Cura 3.10, progress bar doesn't update at all.

bobbanks, 2018-01-30:
Looks like a server restart solved the problem.

closed 2018-01-30 by bobbanks

cesarvandevelde, 2018-01-30:
Good to know! I'll do some experimenting to see if I can reproduce this behavior. If so, I'll add the `RestartNeedingPlugin` mixin to the plugin. That should then resolve the issue for everyone.

benlye, 2018-01-30:
Note that there is also an issue in the latest/current Marlin 2.0.x  where `P` parameters are silently ignored...
https://github.com/MarlinFirmware/Marlin/pull/9393
https://github.com/MarlinFirmware/Marlin/issues/9366

bobbanks, 2018-01-30:
Thanks guys.  This was on a build of 1.1.8.