cesarvandevelde/OctoPrint-M73Progress #4: Progress Bar

luckyheiko, 2018-01-31:
hello

i install the plugin on the octoprinz and change the firmware (marlin 1.1.8)
but i can not see the progress bar.
and also i can not see some m73 comandes in the terminal.

can you please help me?

p.s. sd support is deaktivate

cesarvandevelde, 2018-01-31:
Try entering `M73 P25` into the console and see if that updates the LCD progress bar. If not, the issue is in firmware...

reggaemanu, 2018-01-31:
Seems to be fixed when activating sd_support, but it takes a lot of memory for nothing, I don't even have an sdcard slot. Anyway, that worked on one of my printers just by activating that feature (I've tried to put LCD_SET_PROGRESS_MANUALLY out of the sdsupport IF but that doesn't work) I didn't investigate more to see if that could be possible to activate that function without the sd support

cesarvandevelde, 2018-02-01:
It seems that you are correct. `LCD_SET_PROGRESS_MANUALLY` requires `SDSUPPORT` in order to work. OctoPrint-M73Progress is working as intended, this is a shortcoming in Marlin (MarlinFirmware/Marlin#9178). For now, I'll add a notice to the project's readme to clarify the situation.

closed 2019-12-24 by cesarvandevelde