cesarvandevelde/OctoPrint-M73Progress #7: Wrong behaviour when gcode has M73 command (from Slic3r)

Mr-Groch, 2019-03-26:
Slic3r gcode already has M73 commands, despite this M73Progress plugin still is sending M73 commands . This leads to silly behaviour on printer LCD, especially on beginning of the print, when OctoPrint estimations and Slic3r estimations are quite different.

I suggest that M73 plugin should check if there are M73 commands in gcode, and if they are - plugin should not send own M73 commands.

cesarvandevelde, 2019-03-27:
I haven't heard of this situation yet; I'd guess most users would either configure their slicer or their Octoprint server to output M73 commands.

- What is your use case? I.e. why do you have files with and files without M73 commands?
- Does Slic3r output M73 commands by default?
- Why do you think that G-code should take precedence over Octoprint, and not the other way around?

I wrote this plugin to scratch a personal itch, so it's missing some niche features like this.
Technically, it's definitely possible to implement this by using `protocol_gcodephase_hook` to set a flag, though it's a low-priority feature for me.
I'm open to a pull request though, if you want to have a go at it.

Mr-Groch, 2019-03-27:
- Mainly I use Slic3r, but for some models it is better to use Cura (tree supports, pause at height, speed of slicing) and Cura does not add M73 commands
- This is printer configuration option, default is disabled (I have enabled, becouse I can see that Slic3r estimated progress is better than file size progress)
- Becouse as I mentioned earlier Slic3r progress estimation is better that file size/printed size progress. Also there is an addon PrintTimeGenius for OctoPrint that can make use from M73 command from gcode to inject better estimations to OctoPrint status sidebar. Unfortunately this does not change default printed size/file size progress behaviour for other plugins (like your M73Progress plugin). So I mainly need that plugin only for Cura sliced files.

cesarvandevelde, 2019-03-27:
I'm also running PrintTimeGenius, but had not noticed the discrepancy yet.
I did some digging in the Octoprint source code, and you're right! Progress plugins only receive progress data via the default file size estimation, even when other estimation plugins are installed. 
To me, the more interesting solution would be to find a way to use the PrintTimeGenius estimate, if available...

alnavasa, 2020-04-14:
Is there any way I can implement this to my machine,
By installing the plugin it does nothing, but you say there needs to be some coding done, 
does that coding need to be done when compiling the firmware, or at the beginning of each prints G-Code
I use this firmware builder: https://bultimaker.bulles.eu

cesarvandevelde, 2020-04-17:
@alnavasa Your firmware needs to support the M73 g-code command (hence the name); but not all firmwares support this. For Marlin, `LCD_SET_PROGRESS_MANUALLY` needs to be enabled. 

I'm closing this issue; the original remarks have been largely addressed in v0.2.0, and firmware configuration help is outside the scope of this project...

closed 2020-04-17 by cesarvandevelde