cesarvandevelde/OctoPrint-M73Progress #9: Add remaining time parameter to output gcode

drzejkopf, 2019-11-23:
Hi, is there any chance that M73 R parameter (remaining time) support would be added to this plugin? It is now supported by at least 2 firmwares (Prusa Firmware and Marlin 2.0, see https://github.com/MarlinFirmware/Marlin/pull/15549) and I think it would be a cool feature to have it supported, especially when combined with PrintTimeGenius

cesarvandevelde, 2019-11-27:
Hi drzejkopf. I've spent an evening looking at this, and it seems doable. I got the bare-bones functionality working, but I'll need a couple more days to clean everything up. I'm currently running an older version of Marlin and I'm not too keen on messing with the firmware. Would you be willing to do some beta-testing for this?

drzejkopf, 2019-11-27:
@cesarvandevelde sure I can, I print daily so I can do some testing, and I'm next to printer most time so I can check frequently if everything works fine

cesarvandevelde, 2019-12-01:
@drzejkopf I've just finished a version that is ready for testing: [feature/time_remaining](https://github.com/cesarvandevelde/OctoPrint-M73Progress/archive/feature/time_remaining.zip).
To install, you will probably have to uninstall the current version first. After that, you can install the zip file using the "... from an uploaded archive" option inside the OctoPrint plugin manager.

This version includes two new features:

- Output time left: Outputs the remaining time (in minutes) using the `P` parameter.
- Use time estimate: By default, the plugin uses the progress estimate provided by OctoPrint. With this option, progress is calculated using the following formula: `progress = time_elapsed / (time_elapsed + estimated_time_remaining)`. This matches the way PrintTimeGenius works.

Both features can be enabled/disabled in Octoprint settings > Plugins > M73 Progress. I've done some quick test prints, and everything seems to be working correctly on my end. It would be really great if you could do more in-depth testing on your setup!

drzejkopf, 2019-12-01:
@cesarvandevelde I've just installed plugin (removed old and installed new one from package), but settings page is blank, have you got any idea what's going on? (I've got OctoPrint 1.3.12 on OctoPi 0.16.0, RPi 3B+)
[image]

cesarvandevelde, 2019-12-01:
This may sound silly, but did you try scrolling up? I noticed on my system that the left-hand side and right-hand side scroll together, and that can obscure the two checkboxes...

drzejkopf, 2019-12-01:
@cesarvandevelde Nope, still nothing, I've also rebooted OctoPrint, then rebooted whole RPi and still nothing. I inspected it using "Inspect element" option in Firefox and it seems to be blank too
[image]

cesarvandevelde, 2019-12-01:
Strange... 

- Can you post your octoprint.log?
- Can you check the contents of `~/oprint/lib/python2.7/site-packages/octoprint_m73progress/templates`?

drzejkopf, 2019-12-01:
I attach log and file which was present in given directory (it's the only file that was there)
[files.zip](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/3908765/files.zip)

cesarvandevelde, 2019-12-01:
The log file does not show any errors, and the template file is correct. I've done an install-from-zip on a blank instance of OctoPrint, and that worked correctly as well... Have you tried clearing your browser cache?

drzejkopf, 2019-12-05:
@cesarvandevelde Hi, I have done ~10 prints with new plugin, everything seems to be working properly. Also, clearing browser cache helped with blank settings page

cesarvandevelde, 2019-12-24:
The new version has been released. Thanks for the help!

closed 2019-12-24 by cesarvandevelde