cesarvandevelde/OctoPrint-M73Progress #10: The plugin causes an exception due to missing _progress

poikilos, 2020-01-26:
```
2020-01-24 16:24:49,230 - octoprint.plugin - ERROR - Error while calling plugin m73progress
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_m73progress/__init__.py", line 49, in on_event
    self._progress.reset()
AttributeError: 'M73progressPlugin' object has no attribute '_progress'
```

cesarvandevelde, 2020-01-27:
How frequently do you get this error? Does it always happen, or only sporadically?

poikilos, 2020-01-29:
It appears to have only happened once. Therefore, it may be something odd having to do with refreshing the page while the OctoPrint interface is initializing, but that's only a guess. I couldn't reproduce it by manually doing that. The octoprint version is 1.3.12 and the plugin version is 0.2.0 from pip

amittel, 2020-03-29:
I installed the plugin via the plugin manager yesterday. I kind of have the same error
```
2020-03-29 08:39:40,728 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: SpringFrameFront_FilamentWorld-1.75.gcode, owner: _admin, user: _admin
2020-03-29 08:39:40,742 - octoprint.plugin - ERROR - Error while calling plugin m73progress
Traceback (most recent call last):
  File "/home/octopi/OctoPrint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/octopi/OctoPrint/local/lib/python2.7/site-packages/octoprint_m73progress/__init__.py", line 49, in on_event
    self._progress.reset()
AttributeError: 'M73progressPlugin' object has no attribute '_progress'
2020-03-29 08:39:40,751 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
```
I had no progress bar during the print. At the end it switched to a full progress bar.

---

OctoPrint 1.4.0
Python 2.7
Pluginversion 0.2.0

amittel, 2020-03-30:
I made some more prints and I can confirm, that at the beginning of every print I have the same error.
```
2020-03-30 08:43:01,575 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: MainSpring_FilamentWorld-1.75.gcode, owner: _admin, user: _admin
2020-03-30 08:43:01,597 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-03-30 08:43:01,609 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: MainSpring_FilamentWorld-1.75.gcode, owner: _admin, user: _admin
2020-03-30 08:43:01,621 - octoprint.plugin - ERROR - Error while calling plugin m73progress
Traceback (most recent call last):
  File "/home/octopi/OctoPrint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/octopi/OctoPrint/local/lib/python2.7/site-packages/octoprint_m73progress/__init__.py", line 49, in on_event
    self._progress.reset()
AttributeError: 'M73progressPlugin' object has no attribute '_progress'
2020-03-30 08:43:01,639 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-03-30 08:43:01,656 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
```

cesarvandevelde, 2020-04-02:
@amittel Have you restarted OctoPrint since installing the plugin? 

The `_progress` object is created right [after octoprint startup](https://github.com/cesarvandevelde/OctoPrint-M73Progress/blob/dc063a3e183e09a4929a13c77ff9ad53fda031fd/octoprint_m73progress/__init__.py#L35) and gets used during print events. It doesn't make sense to me that the plugin would receive a `PRINT_STARTED` event before the `on_after_startup` event. Unless if the plugin is used right after installation, maybe (hence my restart theory).

amittel, 2020-04-03:
I investigated a little further and can tell that it has nothing to do with a restart.

I did a complete new venv install mit py3.7 and set it up. I've made a print. Got the error. Now doing a second print without restarting or anything. Just hitted print. Same error. So I have no clue.

Btw: I have no inputform under settings for the plugin. Maybe there is a connection?
[image]

cesarvandevelde, 2020-04-03:
The blank settings page is due to a browser caching issue. Clear your browser cache and it should work. This part is handled by OctoPrint itself, so I don't think I can fix it from my side.

As for the main issue: the root cause is that the `_progress` object gets used before it is created. This object is created in the `on_after_startup()` method, which is called by OctoPrint during startup.

If my theory is correct, the following sequence should cause the error:

- *Starting from a blank install*
- Start OctoPrint
- Install the plugin
- Start a print --> Error
- Start a second print --> Error

The following sequence should not cause an error:
- *Starting from a blank install*
- Start OctoPrint
- Install the plugin
- Stop OctoPrint
- Start OctoPrint
- Start a print --> Should work fine

amittel, 2020-04-04:
Ok. I got it running.
My problem was not realted to the plugin. Restarting Octoprint was not working proberly (had to fix/config pam.d). After fixing it and restarting and printing, the progressbar works now.
Thanks for your help!

cesarvandevelde, 2020-04-04:
I've made a small change so that you're prompted to restart OctoPrint after installation: [bugfix/restart](https://github.com/cesarvandevelde/OctoPrint-M73Progress/archive/bugfix/restart.zip)

Can you try it out before I push the update?

gizmocuz, 2020-06-23:
@cesarvandevelde , I did not had the crash before and was just browsing the issues before I was considering installing this plugin.
I can confirm that your bugfix/restart is asking to restart OctoPrint, and succeeded
Made a test print, and the progress bar is updating beautifully!
Thanks for this plugin ! I wish I found this months ago ;)

cesarvandevelde, 2020-06-24:
Thanks for the heads-up, @gizmocuz! I've pushed out the fix in a new release. Hopefully, that's the end of that. :)

closed 2020-06-24 by cesarvandevelde