cesarvandevelde/OctoPrint-M73Progress #12: Prusa Firmware Improve M0/M1/M117

3d-gussner, 2021-02-23:
Hi,
Prusa-Firmware merged https://github.com/prusa3d/Prusa-Firmware/pull/3034
As you are using the `M73` in your Octoprint Plugin please verify if it will still work after the new Prusa-Firmware release.

cesarvandevelde, 2021-02-23:
I appreciate the heads-up, but I do not own any Prusa3D products. Consequently, I will not do any testing against this firmware release. Pull requests are welcome, provided they retain compatibility with Marlin.

3d-gussner, 2021-02-23:
You can try https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_MK404 if you are on Linux.
Using `./PF-build.sh -m 1 -g 1` will
- Download/prepare Prusa-Firmware build environment
- Guide you to choose correct Prusa-Firmware

After compiling the Prusa-Firmware it will
- Download/prepare MK404 https://github.com/vintagepc/MK404
- Start automatically the MK404 simulator with a serial `/tmp/simavr-uart0` you can use to connect with Octoprint

I also have a WIP Prusa Marlin branch if you want to test Marlin https://github.com/3d-gussner/Marlin/tree/MK3s_2.0.x

3d-gussner, 2021-02-23:
Sorry posted wrong PR number