cesarvandevelde/OctoPrint-M73Progress #13: No M73 updates, an hour into print.

andritolion, 2021-07-30:
Even an hour into the print, there doesn't seem to have been any M73 updates sent to the printer, from what I could tell from watching the terminal. Any idea why that might be? Here's the last 1500 lines that I watched in the OctoPrint terminal, looking for any M73s.

OctoPrint 1.6.1, OctoPi 0.18.0, running on Raspberry Pi 4 Model B Rev 1.4

[No M73 log.txt](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6906704/No.M73.log.txt)

andritolion, 2021-07-30:
Done some investigating, and a quick 15 minute print shows the M73 info very soon after starting. Could it be that the print (which is 13 hours long) is too large for the plugin to handle? (IMO, updated M73 info should be sent every 30 seconds or so... Or whenever it changes or when it loads for the first time.)

cesarvandevelde, 2021-07-30:
The plugin does not decide when the progress bar is updated. It responds to the `on_print_progress()` hook, which is triggered by OctoPrint itself. Long prints should not be a problem, as long as you're not printing from the SD-card.

- Did you restart OctoPrint after installation?
- Has the problem happened again since?
- Are you using the `progress_from_time` setting? (i.e. in combination with PrintTimeGenius)

andritolion, 2021-07-30:
- I have restarted OctoPrint and the whole machine several times as a part of routine testing on my end.
- Yes, the problem still occurs when beginning a 14 hour long print. However, I have noticed that during some time I was sleeping, the time and percent came on and was sent to the printer. Not good timing though.
- Yes, I am using the `progress_from_time` setting with this plugin. The ability this plugin has to show percent and values that are closer to the number PrintTimeGenius provides is important to me. I have briefly used the Detailed Progress plugin (in conjunction with the DisplayLayerProgress plugin with all features that output M73 disabled, that plugin was only used for layer progress, height progress, and time to filament change data in Detailed Progress), but from what I could tell, it did not use PrintTimeGenius percentages or time remaining, so I disabled the M73 features in Detailed Progress (only used it for the M117 ticker) and replaced the features with this plugin. Detailed Progress did immediately show the time remaining and frequently update the percentage, however.

cesarvandevelde, 2021-07-31:
Octoprint always uses the builtin progress estimation algorithm for the `on_print_progress()` hook, even if a more accurate estimate (i.e. PrintTimeGenius) is available. Why it does this I don't know, but my workaround involves calculating progress based on the elapsed time and the total time. This behavior is enabled through the `progress_from_time` option.

My guess is that there is an edge case where `on_print_progress()` gets called before the PrintTimeGenius estimate is computed. If this is the case, it would only happen for longer prints and more complex prints. It would not occur the second time the same file is printed, as estimates are cached by PrintTimeGenius. I tend to print smaller (< 8h) mechanical parts, so that could explain why I have not run into the issue you're describing.

Can you check the following, preferably with related plugins (DisplayLayerProgress, Detailed Progress) disabled?

- The issue should not occur for small prints
- The issue should occur the first time a complex file gets printed, but not the second time (due to caching)
- Were there exceptions logged to `octoprint.log`?

andritolion, 2021-08-02:
Apologies for the late reply. Been recovering from a lack of sleep recently.
- Yes, I can confirm that the issue is not present on short prints.
- This issue has still re-occurred, even after the same model was printed multiple times.
- No, I don't believe there were any exceptions logged to octoprint.log, but feel free to inspect it if you like.

These tests were performed with any plugins that could interfere disabled. I have run the test for an hour and 10 minutes, the whole time, the Remaining field unfilled. Attached are some relevant photos and a copy of my octoprint.log.

[octoprint.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6916107/octoprint.log)
[image]
[image]
[image]

cesarvandevelde, 2021-08-05:
Sorry for the wait. I gave your last log a closer look, and it shows that DisplayETAPlugin is throwing a bunch of errors. Maybe that's causing issues downstream. Can you try disabling it?

I've also prepared a test version that includes debug logging and extra checks for potential edge cases:
[octoprint_m73progress_debug.zip](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6940252/octoprint_m73progress_debug.zip)

To install, uninstall the existing plugin first, then upload the zip file in the OctoPrint plugin manager. This version will output a `plugin_m73progress.log` file. Can you reproduce the issue and then add the Octoprint log and the plugin log here?

andritolion, 2021-08-06:
Here are the requested logs along with a couple pictures.

[octoprint.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6943297/octoprint.log)
[plugin_m73progress.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6943298/plugin_m73progress.log)

[image]
[image]

andritolion, 2021-08-08:
It seems to be quite inconsistent. On this 17 hour print, longer than the other file, it shown the time remaining on the printer display within 10 minutes.
[image]

[image]

cesarvandevelde, 2021-08-13:
Sorry for the late reply; paid work took priority. I took a closer look at your logs, and it seems that `on_print_progress()` never gets called. I've prepared another debug version to check this: [octoprint_m73progress_debug2.zip](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6981110/octoprint_m73progress_debug2.zip). Best to delete the existing logs (`octoprint.log` and `plugin_m73progress.log`) before trying, so that nothing gets mixed up.
 
If this is the case, the problem is caused outside of the plugin. The root of the problem might be a conflicting plugin, so the next step would be to try with all other plugins disabled.

andritolion, 2021-08-13:
No worries! Life happens. I have disabled *all* third-party plugins minus M73 Progress and PrintTimeGenius, and the issue can still be reproduced. Here's a copy of the log files.

[octoprint.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6982748/octoprint.log)
[plugin_m73progress.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6982749/plugin_m73progress.log)
[image]
[image]
[image]

cesarvandevelde, 2021-08-13:
Well, the logs confirm that `on_print_progress()` is not getting called, so the root cause must lie outside of the plugin. I think this issue may have to get kicked up to OctoPrint itself. Out of curiosity, can you try disabling PrintTimeGenius as well? The plugin should fall back on OctoPrint's built-in time estimate.

andritolion, 2021-08-13:
Alright. I'm disabling PrintTimeGenius and will get back to you in an hour.

andritolion, 2021-08-13:
Immediately after starting the print, the time remaining is set.

andritolion, 2021-08-13:
However, it doesn't seem like the time remaining info is actually updated once set. Nearly an hour into print, and it still says 11:55 remaining.
[octoprint.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6983324/octoprint.log)
[plugin_m73progress.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6983325/plugin_m73progress.log)

[image]

[image]

cesarvandevelde, 2021-08-14:
Yup, exactly the same situation as before: `on_print_progress` does not get called.

I've created a minimal plugin that does nothing more than write the progress to a log file: [OctoPrint-ProgressTest.zip](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/6986265/OctoPrint-ProgressTest.zip). If the problem can be reproduced with Octoprint + this plugin (no other 3rd party plugins), then the next step would be opening an issue in the Octoprint repository.

antonioreyna, 2021-08-16:
is it possible to enable the `//#define LCD_SET_PROGRESS_MANUALLY` on Ender3? I tried but got this error

```
 #error "LCD_PROGRESS_BAR requires a character LCD."
```

andritolion, 2021-08-16:
> is it possible to enable the `//#define LCD_SET_PROGRESS_MANUALLY` on Ender3? I tried but got this error
> 
> ```
>  #error "LCD_PROGRESS_BAR requires a character LCD."
> ```

You need custom firmware in order to be able to have progress updates on the Ender 3v2. Check out Jyers firmware. https://github.com/Jyers/Marlin/

Also, sorry for the delay, Cesar. With classes and everything starting up now, I've had limited time to test the minimal plugin, or print anything in general. I'll try to install it when I get home.

andritolion, 2021-08-21:
Sorry for the delay. Never got a chance to test it until today. I was able to reproduce the issue with only the plugin provided. 
[plugin_progresstest.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/7025966/plugin_progresstest.log)
[octoprint.log](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/7025967/octoprint.log)
[image]
[image]
[image]

cesarvandevelde, 2021-08-22:
No worries about the delay, and thanks for trying out the many test scenarios. The last test shows that the root cause of the bug is OctoPrint itself, not one of the plugins. I think it's best to open a new issue in the OctoPrint repo with a link to this discussion. If you want, I'll write up the issue, just let me know...

andritolion, 2021-08-22:
Alright. I'll also attach the gcode file in both issues to see if someone else can repro it. (Packed in a ZIP, else GitHub complains.)
[CE3_Display_holder_Ender_3V2_without_logo.zip](https://github.com/cesarvandevelde/OctoPrint-M73Progress/files/7027479/CE3_Display_holder_Ender_3V2_without_logo.zip)

jneilliii, 2021-12-26:
@cesarvandevelde I did some testing on this for @andritolion and we suspect the issue is with the on_print_progress but not because it's not working, because it is. The issue is that even with PTG installed, the on_print_progress callback is still done on a file size based progress estimate for the increments, not the time based one of PTG. I see that you are calculating the time based progress in your callback, but that callback only gets called on the internal file based increment. PTG overrides the progress bar in the UI, not in the server side code. One way you could get around this is instead of using on_print_progress use your own [RepeatedTimer ](https://docs.octoprint.org/en/master/modules/util.html#octoprint.util.RepeatedTimer) thread and then use sef._printer.get_current_data().

cesarvandevelde, 2021-12-27:
That's exactly right. The `on_print_progress` hook gets called every 1% increment of the file progress, but the plugin calculates the progress value based on the PTG time estimate. 

The root cause of the issue is that `on_print_progress` does not get called at all. In the case of the 17h print, the first update should occur after 1% file progress (roughly `17 * 60 * 0,01 = 10min`). However, the report says that nothing happened after 1 hour had elapsed... To me, this suggests that the problem is caused by an OctoPrint bug.

jneilliii, 2021-12-27:
I ran the file for quite some time with the virtual printer and it did increment though.

andritolion, 2021-12-27:
IMO, running a RepeatedTimer thread that updates the info every 60 seconds would be a better implementation, as even though the percentage isn't changing, the time remaining countdown would be kept up to date. Additionally, the info should also be sent immediately at print start, so that the time estimate will be immediately visible upon starting the print, and not after a minute.

cesarvandevelde, 2021-12-27:
Thanks for clarifying; I was under the impression that the hook was not getting called at all. I agree that a periodic timer implementation would be better, but I can't make make any promises w.r.t. timeframe. Pull requests are welcome though, in case someone wants to have a crack at it...

ProtomakerSprint, 2023-03-08:
I am having a similar issue of the M73 plugin not displaying the progress. Just an empty bar:

It is showing E: and R: (elapsed and remaining) times on my : BBT TF35 Dual display which I use as the  LCD12864

Configuration_adv.h, which was working, as I normally use, modified to suit update for  Marlin-bugfix-2.1.2
#define SET_PROGRESS_MANUALLY  // Protomaker Sprint Show Progress using OctoPrint and M73 Plugin

It was working before

So far as I am aware it stopped working when I updated to
 
 Marlin-bugfix-2.1.2, this changed the LCD12864 lay out a bit

OctoPrint 1.8.7 Python 3.9.2 OctoPi 1.0.0

Above updated to this

Thanking you

ProtomakerSprint, 2023-03-09:
> LCD_PROGRESS_BAR

Have you the correct printer set Configuration_adv.h says:-

 #if EITHER(HAS_MARLINUI_HD44780, IS_TFTGLCD_PANEL)
    //#define LCD_PROGRESS_BAR 

in my case in Configuration.h I use and leave the above uncommitted

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER // Protomaker Sprint TFT35 Dual display

ProtomakerSprint, 2023-03-09:
> Have you the correct printer set Configuration_adv.h says:-
>
> #if EITHER(HAS_MARLINUI_HD44780, IS_TFTGLCD_PANEL) //#define LCD_PROGRESS_BAR
>
> in my case in Configuration.h I use and leave the above [...]

LCD_PROGRESS_BAR

Have you the correct printer set Configuration_adv.h says:-

#if EITHER(HAS_MARLINUI_HD44780, IS_TFTGLCD_PANEL)
//#define LCD_PROGRESS_BAR

in my case in Configuration.h I use and leave the above uncommitted

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER // Protomaker Sprint TFT35 Dual display

cesarvandevelde, 2023-03-11:
@ProtomakerSprint Your issue is unrelated to this discussion, and the cause is likely outside this plugin.

That being said, please refer to the [Marlin documentation for M73](https://marlinfw.org/docs/gcode/M073.html):

* `LCD_SET_PROGRESS_MANUALLY` is required.
* `USE_M73_REMAINING_TIME` since v2.0.0
* Test by sending `M73 P25 R43` manually.

ProtomakerSprint, 2023-03-16:
I am using Marlin; my guide to gcode says M73 not available, set percentage build
M73 P25 R43 manually did noting at all, no change on display
------

SanityCheck.h states:-
  #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
--------

in Configuration_adv.h I have:-
// Add the G-code 'M73' to set / report the current job progress
#define SET_REMAINING_TIME true  // Protomaker Sprint SET_PROGRESS_PERCENT, SET_REMAINING_TIME, SET_INTERACTION_TIME to be enabled."

#define SET_PROGRESS_MANUALLY  // Protomaker Sprint Show Progress using OctoPrint and M73 Plugin
#if ENABLED(SET_PROGRESS_MANUALLY)
  //#define SET_PROGRESS_PERCENT          // Add 'P' parameter to set percentage done, otherwise use Marlin's estimate
  //#define SET_REMAINING_TIME            // Add 'R' parameter to set remaining time, otherwise use Marlin's estimate
  //#define SET_INTERACTION_TIME          // Add 'C' parameter to set time until next filament change or other user interaction
  #if ENABLED(SET_INTERACTION_TIME)
    #define SHOW_INTERACTION_TIME         // Display time until next user interaction ('C' = filament change)
  #endif
  //#define M73_REPORT                    // Report progress to host with 'M73'
#endif
======
Thanking You

ProtomakerSprint, 2023-03-16:
Formatting got a bit mucked up above; sorry

ProtomakerSprint, 2023-03-16:
P.S. this bar used to show, similar Marlin setting so far as I know

cesarvandevelde, 2023-03-17:
Again: this is not a plugin issue but a Marlin configuration issue. Here are [the relevant settings](https://github.com/MarlinFirmware/Marlin/blob/89e8257008507821543af8811ab78ef0365b2741/Marlin/Configuration_adv.h#L1494-L1513). Uncomment `//#define SET_PROGRESS_MANUALLY` and make sure the other settings are at their default values. Test by sending `M73` commands manually.

If that does not work, ask your printer manufacturer for support.

ProtomakerSprint, 2023-03-22:
**M73 progress bargraph working now**
**A Massive THANK YOU to [ellensp](https://github.com/ellensp) commented** https://github.com/MarlinFirmware/Marlin/issues/25543#issuecomment-1475635093 •

A Video Short shows it working, **_It was My CODE_**
https://youtube.com/shorts/A1-cIO4NQMg?feature=share

I changed the Code to the working code that has ****:-
M73 Plugin
#if ENABLED(SET_PROGRESS_MANUALLY)
//#define SET_PROGRESS_PERCENT // Add 'P' parameter to set percentage done, otherwise use Marlin's estimate.Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
#define SET_REMAINING_TIME // Add 'R' parameter to set remaining time, otherwise use Marlin's estimate. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
//#define SET_INTERACTION_TIME // Add 'C' parameter to set time until next filament change or other user interaction
#if ENABLED(SET_INTERACTION_TIME)
#define SHOW_INTERACTION_TIME // Display time until next user interaction ('C' = filament change)
#endif
//#define M73_REPORT // Report progress to host with 'M73'
#endif